jsreynaud/kicad-action-scripts #5: Causes pcbnew to segfault

Phara0h, 2018-06-14:
With the latest nightly build, it causes kicad pcbnew to segfault. I honestly don't know enough about wx to try to fix the issue on my own. So I figured I would tell you in hopes you can help.

nickoe, 2018-07-05:
https://bugs.launchpad.net/kicad/+bug/1762432

herostrat, 2018-07-25:
Same here
I am thinking to file a bugreport, but I don't know for sure if it's right.
Could it be because KICAD_SCRIPTING_WXPYTHON=OFF

It worked on the rc3 though...

----

> Application: kicad
> Version: 5.0.0-fee4fd1~65~ubuntu18.04.1, release build
> Libraries:
>     wxWidgets 3.0.4
>     libcurl/7.58.0 OpenSSL/1.1.0g zlib/1.2.11 libidn2/2.0.4 libpsl/0.19.1 (+libidn2/2.0.4) nghttp2/1.30.0 librtmp/2.3
> Platform: Linux 4.15.0-29-generic x86_64, 64 bit, Little endian, wxGTK
> Build Info:
>     wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 2.24
>     Boost: 1.65.1
>     OpenCASCADE Community Edition: 6.9.1
>     Curl: 7.58.0
>     Compiler: GCC 7.3.0 with C++ ABI 1011
> 
> Build settings:
>     USE_WX_GRAPHICS_CONTEXT=OFF
>     USE_WX_OVERLAY=OFF
>     KICAD_SCRIPTING=ON
>     KICAD_SCRIPTING_MODULES=ON
>     KICAD_SCRIPTING_WXPYTHON=OFF
>     KICAD_SCRIPTING_ACTION_MENU=ON
>     BUILD_GITHUB_PLUGIN=ON
>     KICAD_USE_OCE=ON
>     KICAD_USE_OCC=OFF
>     KICAD_SPICE=ON
>

jsreynaud, 2018-07-25:
It can't work on Ubuntu 18.04 yet. Wxpython is not compatible with kicad.

Regards

closed 2019-05-23 by jsreynaud