jsreynaud/kicad-action-scripts #7: BuildFilledSolidAreasPolygons does not exist anymore

m3x1m0m, 2018-11-25:
Hi :) Cool scripts, you have here! Chapeau! 

The via stitching plugin did not work out of the box for me. I compiled Kicad maybe three weeks ago from the master branch. I realized, that the function for refilling the copper areas does not exist any more. Do you know of any replacement? Perhaps that one **BuildSmoothedPoly.**
 
https://github.com/jsreynaud/kicad-action-scripts/blob/f65c507ddcacdfc50b7f05f2119f4d5ad34a88bd/ViaStitching/FillArea.py#L233

Also there are some minor things I changed to get it running. Just a hotfix. 

https://github.com/m3x1m0m/kicad-action-scripts

W-M-D, 2019-03-19:
Hey your branch doesn't work with kicad 5, any ideas on how to fix. I'm about to dig into this.

jsreynaud, 2019-03-20:
Yes I have to review this on KiCad 5.1... I'll work on it on next weeks.

closed 2019-05-23 by jsreynaud