jsreynaud/kicad-action-scripts #9: ViaStiching won't register in menu

karrots, 2019-05-12:
I've copied the directory to the same place as other scripts. It doesn't want to register. I'm running Kicad 5.1.2 on Ubuntu Linux. If directed where to maybe find a stack trace more than happy to get more information.

jsreynaud, 2019-05-23:
Could share your version information on KiCad (Menu Help => About KiCad => Copy Version info)

Thanks,

karrots, 2019-05-24:
Application: kicad
Version: 5.1.2-f72e74a~84~ubuntu18.10.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.61.0 OpenSSL/1.1.1 zlib/1.2.11 libidn2/2.0.5 libpsl/0.20.2
(+libidn2/2.0.4) nghttp2/1.32.1 librtmp/2.3
Platform: Linux 4.18.0-18-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.67.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.61.0
    Compiler: GCC 8.2.0 with C++ ABI 1013

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

jsreynaud, 2019-05-27:
Could you retry with last commt. It should fix the issue...

karrots, 2019-05-27:
commit c3f4dfbc25fd4e66c701f80cef0cb4cbabd1d763 does seem it fix the issue.

closed 2019-05-27 by karrots