jsreynaud/kicad-action-scripts #10: ViaStichting - nothing happens

Munken, 2019-06-03:
Hi Reynaud

I have installed your plugin in the latest nightly version.  
I placed the ViaStiching folder in ~/.kicad_plugins. It registers fine, 
but when I select run nothing happens.

Got any idea of how one could dig deeper into this?

My version string is this
```
Application: KiCad
Version: 6.0.0-unknown-27d097a~100~ubuntu18.04.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.58.0 OpenSSL/1.1.0g zlib/1.2.11 libidn2/2.0.4 libpsl/0.19.1 (+libidn2/2.0.4) nghttp2/1.30.0 librtmp/2.3
Platform: Linux 4.15.0-50-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.22
    Boost: 1.65.1
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.58.0
    Compiler: GCC 7.4.0 with C++ ABI 1011

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON
```

W-M-D, 2019-07-04:
Yeah I have been digging through it and it seems that the entire script is broken it hangs in area.HitTestForEdge and if you skip that it hangs further down the line. Just run kicad in terminal mode and you can see debugging output

W-M-D, 2019-07-04:
ok i was able to fix it. i'm going to fork this and submit a patch

dom11990, 2019-08-10:
@W-M-D I tried your patch but I still seem to have the hanging behavior and then nothing happens. Any idea I'm running Kicad 5.1 on Windows so I haven't the slightest clue on how to see debug information after I've started the program. Any idea on what could be wrong? I'll try it on linux at some point in the next days to see if the problem is present there as well. Hopefully I'll have more useful information about the problem then.

weirdgyn, 2019-09-25:
Hallo I did a check today and the code execution doesn't get below line 49 (file `FillAreaAction.py`):
`        if modal_result == wx.ID_OK:`

If I may suggest you should get vias size/drill/clearance values from pcbnew settings instead insert static values that may not work. Net name and checkbox 'Only under selected zone' could be retrieved from selected area (if any).

easyw, 2019-09-26:
@weirdgyn 
please have a look at my PR https://github.com/jsreynaud/kicad-action-scripts/pull/12
It should fix 5.1.4 compatibility issues.
I also have added a Net Combo as in your recent plugin.
Maurice

weirdgyn, 2019-09-27:
Wow! Glad to be of any _inspiration_ with my work! I'll check your plugin ASAP.

jsreynaud, 2019-09-27:
PR #12 is merged ! Thanks...

closed 2019-09-27 by jsreynaud