jsreynaud/kicad-action-scripts #13: Crash found (tested on easy-w's compatibility branch)

Shackmeister, 2019-09-26:
Steps to reproduce:
1: Load up the attached board file.
[Via_stich_test.zip](https://github.com/jsreynaud/kicad-action-scripts/files/3660218/Via_stich_test.zip)
2: Run Viastitch plugin
3: set the Via Copper Size to 0.4 (0.5 and below crashes)
4: run the script
5: crash with the following error code
[image]

easyw, 2019-09-27:
@Shackmeister 
it should be fixed now
I have removed original try:except because it was failing without messages... 
Now it should just alert but finishing the job.
https://github.com/jsreynaud/kicad-action-scripts/pull/12/commits/554c74abb455f7e2df22c36526a7c1862804c536
PR https://github.com/jsreynaud/kicad-action-scripts/pull/12 upated

jsreynaud, 2019-09-27:
Last commit fix this...

closed 2019-09-27 by jsreynaud