jsreynaud/kicad-action-scripts #21: ViaStitching placing 0 vias

cnieves1, 2020-01-27:
Hi, 
I am using Kicad 5.1.4 windows version.
When runnig ViaStitching on this simple circular board:

[test1.zip](https://github.com/jsreynaud/kicad-action-scripts/files/4116584/test1.zip)

[image]

I always get "0 vias placed":
11:19:36: Via Stitching: Version 1.5
11:19:36: Processing Target Area: GND, LayerName: F.Cu...
11:19:36: Processing all vias of target area...
11:19:36: Processing all pads...
11:19:36: Processing all tracks...
11:19:36: Processing all existing drawings...
11:19:36: Remove vias to guarantee step size...
11:19:36: 0 vias placed
Done!

If I set "Debug mode" ON, I get an error:

[image]

What can I do to get it running?

TIA

easyw, 2020-01-27:
@cnieves1 
your board has a B.Cu zone assigned to no net... then the script is not allowed to assign vias on that zone.
Everithing will just work fine if you will assign to the B.Cu zone the GND net

cnieves1, 2020-01-27:
Hi,
you were right. Sorry!

I was distilling an easy test case from a more complex board, and I thought I got it, but it was obviously wrong.

Consider this example:
[test1.zip](https://github.com/jsreynaud/kicad-action-scripts/files/4116686/test1.zip)

This has a GND zone in F.Cu, B.Cu, and In2.Cu, and another area for other net in In1.cu.
This is the like the board I'm working on. In my board, In1.Cu would be assigned to 3V3.

So, if I run ViaStitching on this board, I get 0 vias placed. If I change the In1.Cu area to GND, I will get vias placed as expected.
Now I know the workaround: disable the In1.Cu area before running ViaStitching. 

Could it get smarter and place vias without consider other copper areas? when vias are placed, I can remove some of them if I get some problem...

Thanks!

easyw, 2020-01-27:
> Could it get smarter and place vias without consider other copper areas? when vias are placed, I can remove some of them if I get some problem...

I don't understand your issue... would you want the plugins to add vias in DRC violations?
If you want your board stiffed with GND vias, simply add your In1.Cu Fill Zone at the end of your design process.

cnieves1, 2020-01-27:
Let me check... 

I take the above example, with a GND zone in F.Cu, B.Cu, and In2.Cu, and another area for other net in In1.cu.

I add manually 4 vias at random locations. Pcbnew automatically assign them the GND net.
Then I run the DRC. 
Result: no DRC violations.

[image]

If I do the same with ViaStitching, I get 0 vias placed.

Of course I don't want vias placed adding DRC violations, but as this little example showed, there is no DRC violations doing that.

I would expect the same behaviour from Pcbnew and from ViaStitching. 
After this test, I know I can do the job adding the In1.Cu Fill zone at the end, but that is just a workaround. It is not the expected behaviour. 

I know this requires a smarter ViaStitching plugin. 
I don't know the drawbacks neither the difficulty of doing so. 
That's why I am raising this issue...

And another different issue is why I get an error if I run ViaStitching with debug option enabled (see my first post)...

jsreynaud, 2021-07-06:
If you are using 5.1 version of KiCad, could you recheck with the updated version of this plugin on the appropriated branch (Branch named V5.1)

cnieves1, 2021-07-24:
I'm sorry. I tried by got the following (maybe unrelated) error after clicking on the ViaStitching button (no dialog showed).

[image]

What can I do to solve it? I'm using:
Application: Pcbnew
Version: 5.1.8+dfsg1-1~bpo10+1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.64.0 OpenSSL/1.1.1d zlib/1.2.11 libidn2/2.0.5 libpsl/0.20.2 (+libidn2/2.0.5) libssh2/1.8.0 nghttp2/1.36.0 librtmp/2.3
Platform: Linux 5.8.0-0.bpo.2-amd64 x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.67.0
    OpenCASCADE Technology: 7.3.0
    Curl: 7.64.0
    Compiler: GCC 8.3.0 with C++ ABI 1013

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=OFF
    KICAD_USE_OCC=ON
    KICAD_SPICE=ON

jsreynaud, 2021-07-28:
> I'm sorry. I tried by got the following (maybe unrelated) error after clicking on the ViaStitching button (no dialog showed).
>
> [image]
>
> What can I do to solve it? I'm using:
> Application: Pcbnew [...]

It's unrelated. Please open a new issue for that if the error still there.

closed 2021-07-28 by jsreynaud

82Lucas, 2024-09-07:
HI, I am using the script for the very first time together with kiCad V8.0.4 and I am having the issue "Done. 0 vias placed"
I have done the copper fill assigned to GND on 3 layer and merged all of them.
Any advice on what I am missing?
[image]

jsreynaud, 2024-09-07:
Hi,

You have to put two areas connected to GND on top and also bottom. As I see
it is not the case in your screenshot.

82Lucas, 2024-09-08:
@jsreynaud, the GND areas were connected on 1st, 3rd and 4th layers, but the issues was that I had a Vcc area on the 2nd layer, and seems that the script does´n like it. 
[image]
Removing the Vcc area , running the via stitching script and then adding back the Vcc plane did the trick!
[image]

BTW, is there a way to indicate the position of the first via in order of get them cantered in the PCB?
Thank you for the support!