jsreynaud/kicad-action-scripts #23: Via stitching generator ignores Via Grid

MarkJeronimus, 2020-04-06:
Under certain circumstances, whatever you enter in the Via Grid field, it always places vias ~2.54mm apart (the initial value of the field). Under other circumstances it doesn't place any via at all. I can't get a grip on it. It seems to have to do with which pre-defined via I select. e.g. it may succeed on 0.8/0.4mm vias and fail on 0.6/0.3mm vias.

jsreynaud, 2021-07-06:
Could you check with new after commit 3e5e1b5

MarkJeronimus, 2021-07-07:
It still happens. And I don't get why I didn't specify any parameters to reproduce it. Now I have: https://imgur.com/a/POeOEod

jsreynaud, 2021-07-07:
Could you tell me if you are using the V5.1 branch of master branch ?

jsreynaud, 2021-08-24:
Close since there is no answers...

MarkJeronimus, 2021-08-24:
Sorry I haven't been able to get around answering this yet.
Where can I find the version number?
All I know is that I have the version from 3e5e1b5.

jsreynaud, 2021-08-24:
Ok, thanks. Could you tell me what KiCad version you use (5.1.9 or if nightly build the commit you use)

MarkJeronimus, 2021-08-24:
```
Version: 5.1.6-c6e7f7d~87~ubuntu18.04.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.58.0 OpenSSL/1.1.1 zlib/1.2.11 libidn2/2.0.4 libpsl/0.19.1 (+libidn2/2.0.4) nghttp2/1.30.0 librtmp/2.3
Platform: Linux 5.4.0-80-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.22
    Boost: 1.65.1
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.58.0
    Compiler: GCC 7.5.0 with C++ ABI 1011

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON
```
We cannot use nightly or because of requirements of at work.

jsreynaud, 2021-08-24:
Ok, In this case, don't forget to use the v5.1 branch of the plugin. The commit 3e5e1b5 is only for nightly builds.
So please switch on branch v5.1 of the plugin.

closed 2021-09-16 by jsreynaud