jsreynaud/kicad-action-scripts #30: Crash on start ViaStitching plugin

dj140, 2021-02-13:
I‘m using kicad 5.1.5 version
[image]
after start ViaStitching plugin 
[image]

OmarAlkassab, 2021-02-22:
Same error here.

[image]
[image]

josenestorpardoherrera, 2021-02-25:
Same error here. I will have to do it manually :s
[image]

jsreynaud, 2021-03-01:
Are you sure that you get the v5.1 branch of the plugin ? The master branch is for nightly build and I had recheck everything. In V5.1 its working correctly.

OmarAlkassab, 2021-03-02:
> Are you sure that you get the v5.1 branch of the plugin ? The master branch is for nightly build and I had recheck everything. In V5.1 its working correctly.

Yes you are right, it worked perfectly when I used the v5.1 branch of the plugin.
[image]

wojiaodingsan, 2021-03-18:
> Are you sure that you get the v5.1 branch of the plugin ? The master branch is for nightly build and I had recheck everything. In V5.1 its working correctly.

I have the same problem when using the default version, after changed to v5.1 it solved, thanks.

arthurmagill, 2021-04-14:
Same error here. Version info:

Application: KiCad
Version: 5.1.9-73d0e3b20d~88~ubuntu18.04.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.58.0 OpenSSL/1.1.1 zlib/1.2.11 libidn2/2.0.4 libpsl/0.19.1 (+libidn2/2.0.4) nghttp2/1.30.0 librtmp/2.3
Platform: Linux 4.15.0-137-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.22
    Boost: 1.65.1
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.58.0
    Compiler: GCC 7.5.0 with C++ ABI 1011
Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

jsreynaud, 2021-07-06:
@arthurmagill . Please check that the plugin branch you use is V5.1

closed 2021-07-06 by jsreynaud