jsreynaud/kicad-action-scripts #34: Crash on start on 5.1 branch on windows (5.1.10)

rhaamo, 2021-07-28:
Using KiCad 5.1.10 on windows, getting this error when launching the via stitching plugin:
[image]

Apparently the same thing as #16 but the fix don't seems to be present in 5.1 branch.

Manually backporting this section: https://github.com/jsreynaud/kicad-action-scripts/pull/15/files#diff-bae6a893773ecb9a7f9faa4ce6f2d73f4f893d7f273ce0e11c67a5e5f19a478dR22-R26 (import, if/else) seems to fix it.

Kicad:
```
Application: KiCad
Version: (5.1.10)-1, release build
Libraries:
    wxWidgets 3.0.5
    libcurl/7.71.0 OpenSSL/1.1.1g (Schannel) zlib/1.2.11 brotli/1.0.7 libidn2/2.3.0 libpsl/0.21.0 (+libidn2/2.3.0) libssh2/1.9.0 nghttp2/1.41.0
Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.73.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.71.0
    Compiler: GCC 10.2.0 with C++ ABI 1014
Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON
```

jsreynaud, 2021-07-28:
Please see my last commit. It should have fix that.

rhaamo, 2021-07-28:
works fine, thanks !

commit 523525bb (2021-07-28): Fix #34
closed 2021-07-28 by rhaamo