jsreynaud/kicad-action-scripts #36: Python Error in Via Stitching Generator

albtronics, 2021-08-23:
When I open **Via Stitching Generator** Plugin, it shows this error:

[image]
I think its due to the conflicting versions of python. ( _KiCad Windows version's core is running on Python 2.7_ )
Please help.

_My KiCad version(on Windows 10):_

[image]

Also, I have a request, It would be great, if there is a quick **Light/Dark Theme Toggle plug-in.**

Thank you in advance, : )

jsreynaud, 2021-08-24:
As I see you use KiCad 5.1.9 and the version of Via Stitching Generator is "master" branch that is supposed to work only with "dev" version of KiCad.

You have to checkout the "v5.1" branch of Via Stitching Generator to fix that (see README.md).

closed 2021-08-24 by jsreynaud

albtronics, 2021-08-25:
Thank you so much for replying Reynaud. *I didn't see that branch
initially.*

*Please consider making a Light/Dark theme toggler plugin for KiCad.*

*Warm regards*
*Albin Joseph C.R*

jsreynaud, 2021-08-25:
Ok. for making a Light/Dark theme toggler plugin for KiCad, I create a new issue for that.

Regards,