jsreynaud/kicad-action-scripts #42: FillAreaAction.py Error on Windows and Kicad 5.1

alborzs, 2021-09-27:
Hi,
Anyone recognize this error:
[image]

Using [ab13588](https://github.com/jsreynaud/kicad-action-scripts/tree/v5.1#:~:text=1mm%20(fix%3A%20%2341)-,ab13588,-11%20days%20ago)

I am on the (latest 5.1) Kicad version:
```
Application: Pcbnew
Version: (5.1.10)-1, release build
Libraries:
    wxWidgets 3.0.5
    libcurl/7.71.0 OpenSSL/1.1.1g (Schannel) zlib/1.2.11 brotli/1.0.7 libidn2/2.3.0 libpsl/0.21.0 (+libidn2/2.3.0) libssh2/1.9.0 nghttp2/1.41.0
Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.73.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.71.0
    Compiler: GCC 10.2.0 with C++ ABI 1014

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON
```

jsreynaud, 2021-10-07:
Could you try to change line 27 of file ViaStitching/FillAreaDialog.py:
from self.SetSizeHints to self.SetSizeHintsSz

And tell me...

jsreynaud, 2022-04-27:
Close because any update

closed 2022-04-27 by jsreynaud