jsreynaud/kicad-action-scripts #44: cannot generate via stitching

gschelotto, 2021-11-22:
Hi,

I'm trying to run the script with no success. Here's the error I get

[image]

My KiCad version Info is:

```
Application: KiCad
Version: 5.1.12-84ad8e8a86~92~ubuntu20.04.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3
Platform: Linux 5.4.0-90-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.71.0
    OpenCASCADE Technology: 7.5.2
    Curl: 7.68.0
    Compiler: GCC 9.3.0 with C++ ABI 1013
Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=OFF
    KICAD_USE_OCC=ON
    KICAD_SPICE=ON
```

Any hints? I was able to run it some months ago with a prior Kicad version (v5.10 I think)
gaston

Razzorback225, 2021-11-22:
Got the same issue, did you solve it?

jsreynaud, 2021-11-24:
I thing you are using the master branch of the plugin instead of 5.1 branch. Since you are using KiCad 5.1.12, please use branch 5.1... 

Take a look at the README.md file...

gschelotto, 2021-12-03:
Thank you, it's solved.

closed 2021-12-03 by jsreynaud