jsreynaud/kicad-action-scripts #45: TRun Time Error

DOGKiCad, 2021-12-10:
[image]

Any idea what's wrong?

jsreynaud, 2021-12-11:
Please check that you are using the right plug-in 's git branch according
your KiCad version.

DOGKiCad, 2021-12-13:
I had the wrong branch downloaded but I've corrected that but not it not appearing in tools:
[image]

jsreynaud, 2022-03-04:
Please check that you are installing this plugin in the right place. You should be able to open the plugin directory directly from pcbnew menu to be sure that you are using the good directory.

jsreynaud, 2022-04-27:
Close because any update...

closed 2022-04-27 by jsreynaud