jsreynaud/kicad-action-scripts #53: plugin not working with kicad 7

kien242, 2023-02-16:
ViaStitching doesn't seem to work with the latest version of Kicad (Kicad7)

timcooijmans, 2023-02-16:
Dirty fix to get it working:
```diff --git a/ViaStitching/FillArea.py b/ViaStitching/FillArea.py
index 78d295e..0beafe1 100644
--- a/ViaStitching/FillArea.py
+++ b/ViaStitching/FillArea.py
@@ -277,7 +277,7 @@ STEP         = '-'
         for area in all_areas:
             area_layer = area.GetLayer()
             area_clearance = area.GetLocalClearance()
-            area_priority = area.GetPriority()
+            area_priority = area.GetAssignedPriority()
             is_rules_area = area.GetIsRuleArea()
             is_rule_exclude_via_area = area.GetIsRuleArea() and area.GetDoNotAllowVias()
             is_target_net = (area.GetNetname() == self.netname)  # (area.GetNetname().upper() == self.netname)
@@ -290,7 +290,7 @@ STEP         = '-'
                 for dx in [-offset, offset]:
                     # All 4 corners of the via are testet (upper, lower, left, right) but not the center
                     for dy in [-offset, offset]:
-                        point_to_test = wxPoint(via.PosX + dx, via.PosY + dy)
+                        point_to_test = VECTOR2I(wxPoint(via.PosX + dx, via.PosY + dy))
 
                         hit_test_area = area.HitTestFilledArea(area.GetLayer(), point_to_test)             # Collides with a filled area
                         hit_test_edge = area.HitTestForEdge(point_to_test, 1)              # Collides with an edge/corner
@@ -598,7 +598,7 @@ STEP         = '-'
 
                             # Offset is half the size of the via plus the clearance of the via or the area
                             offset = 0  # Use an exact zone match
-                            point_to_test = wxPoint(int(current_x), int(current_y))
+                            point_to_test = VECTOR2I(wxPoint(int(current_x), int(current_y)))
                             hit_test_area = area.HitTestFilledArea(
                                 area.GetLayer(), point_to_test, int(offset))             # Collides with a filled area
                             # Collides with an edge/corner
@@ -764,7 +764,7 @@ STEP         = '-'
                         ran_x = (random.random() * l_clearance / 2.0) - (l_clearance / 4.0)
                         ran_y = (random.random() * l_clearance / 2.0) - (l_clearance / 4.0)
 
-                    self.AddVia(wxPoint(via.PosX + ran_x, via.PosY + ran_y), via.X, via.Y)
+                    self.AddVia(VECTOR2I(wxPoint(via.PosX + ran_x, via.PosY + ran_y)), via.X, via.Y)
                     via_placed += 1
 
         if self.debug:
```

kien242, 2023-02-19:
[image]
sorry, it still not working

CGrassin, 2023-02-21:
I fixed most of the issues with KiCad 7 in PR https://github.com/jsreynaud/kicad-action-scripts/pull/54
While it is not merged, you may want to use my fork  (https://github.com/CGrassin/kicad-action-scripts) to get this plugin running in KiCad 7. You can report the remaining KiCad 7 issues in the PR.
@jsreynaud I would suggest to test, review and merge this PR as-is to get this repository in a working state with KiCad 7, and then we can iron out the remaining issues with later commits.

jsreynaud, 2023-02-21:
@CGrassin  Thanks for you for you PR. I had merge it on the main branch. I create a new branch for tracking modification for kicad7

closed 2023-02-21 by jsreynaud