jsreynaud/kicad-action-scripts #55: Remaining issues with Via Stitching in KiCad 7

CGrassin, 2023-02-21:
This issue is a follow-up to the merging of PR #54 .
There are some remaining issues with the Via Stitching plugin in KiCad 7: the patterns Concentric, Outline and Outline (No Holes) crash the plugin. This is the stack-trace: 
[image]

This is because the function ZONE.RawPolysList has been removed from the KiCad 7 API: https://docs.kicad.org/doxygen-python-7.0/classpcbnew_1_1ZONE.html I am not aware of a replacement...

jsreynaud, 2023-02-22:
Perhaps [iterate](https://docs.kicad.org/doxygen-python-7.0/classpcbnew_1_1ZONE.html#affa63b3e95592a82b573598971c5c13a) method could be a good alternative...

GioCC, 2023-06-08:
Is there any foreseeable fix on this issue?

jsreynaud, 2023-06-08:
Last merge on 7.0 branch fix that.

closed 2023-06-08 by jsreynaud