jsreynaud/kicad-action-scripts #56: "Random it" does not check for clearance

CGrassin, 2023-02-21:
The "random it" function does not properly account for clearance.
Example:
[image]
Result:
[image]

This is because the random offset is not properly bounded.

closed 2023-02-22 by jsreynaud

matthuszagh, 2023-02-28:
I'm experiencing a similar issue except that the vias violate the clearance of other objects like traces.
[image]

matthuszagh, 2023-02-28:
Actually I'm seeing this issue occasionally even if I don't select randomization. For example:

[image]

[image]

CGrassin, 2023-02-28:
@matthuszagh  Good catch, I managed to reproduce it, I am creating a new issue to track it.

PR #57 (2023-02-22): Fix random offset bound (Issue #56)
This PR fixes #56 : the maximal safe distance a via can ever be moved around without risking to bump into another via (including its clearance) is geometrically:
```python
(self.step - (self.clearance + self.size)) / 2
```
This might be negative if the step size is very small, hence the max with 0.