jsreynaud/kicad-action-scripts #61: Segmentation fault kicad 7

guillemgracia, 2023-03-08:
Hi,

When trying to use via stiching on Pop!_OS 22.04 LTS I get a segmentation fault. I attach captures of the debug mode and backtrace.

```
(gdb) bt -full
#0  0x00007f96b70c82f1 in  () at /usr/lib/kicad/bin/_pcbnew.kiface
#1  0x00007f96b6bc1f05 in  () at /usr/lib/kicad/bin/_pcbnew.kiface
#2  0x00007f96b6aa6932 in  () at /usr/lib/kicad/bin/_pcbnew.kiface
#3  0x00007f96b7087473 in  () at /usr/lib/kicad/bin/_pcbnew.kiface
#4  0x00007f96ea4dc2b3 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007f96ea094b43 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
        ret = <optimized out>
        pd = <optimized out>

                      unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140285694688112, 8509343378183445230, 140286079874624, 0, 140286148298832, 140285694688464, -8558377107163292946, -8558362505327387922}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
#6  0x00007f96ea126a00 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81

```

```
Final result:
________________________________________
|                                      |
|                                      |
|                                      |
|                                      |
|                                      |
|                                      |
|                                      |
|                                      |
|           P TPPPPPPPPPPP             |
|           P TPPPPPPPPPPP             |
|            PTTTTTTTTXXXTX            |
|            PPPTTTPPTPPXTT            |
|            PTTTTTXPXPPXTX            |
|            PXPPPTPPXTTXTP            |
|            XTPPPTPPPPTTTP            |
|            XXPPTTPPTTTTTP            |
|            XTXPTTPPTTXXTP            |
|            XXXPPTPPTXTTTX            |
|            XTTPTTPPPPTTTX            |
|            XTPPPTPPXXTXTX            |
|            XXTPTTXKXTXXTX            |
|           P TPPPPPPPPPPP             |
|           P XPPPPPPPPPPP             |
|                                      |
|                                      |
|                                      |
|                                      |
|                                      |
|                                      |
________________________________________

OK           = 'X'
NO_SIGNAL    = ' '
OTHER_SIGNAL = 'O'
KEEPOUT      = 'K'
TRACK        = 'T'
PAD          = 'P'
DRAWING      = 'D'
STEP         = '-'

1678283482.9386125: Line 778
Segmentation fault (core dumped)
```

Thanks for the plugin though!

InkOverSword, 2023-03-15:
I've run into the same issue. Hopefully this is helpful information.

### Version Info
```
Application: KiCad x86_64 on x86_64

Version: 7.0.1, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.13.0
	HarfBuzz 7.1.0
	FontConfig 2.14.2
	libcurl/7.88.1 OpenSSL/3.0.8 zlib/1.2.13 brotli/1.0.9 zstd/1.5.4 libidn2/2.3.4 libpsl/0.21.2 (+libidn2/2.3.4) libssh2/1.10.0 nghttp2/1.52.0

Platform: Arch Linux, 64 bit, Little endian, wxGTK, , x11

Build Info:
	Date: Mar 12 2023 01:48:29
	wxWidgets: 3.2.2 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.81.0
	OCC: 7.6.3
	Curl: 7.88.1
	ngspice: 39
	Compiler: GCC 12.2.1 with C++ ABI 1017

Build settings:
	KICAD_USE_EGL=ON
	KICAD_SPICE=ON
```

### Backtrace
```
Starting plugin CircularZone
Starting plugin ViaStitching
[New Thread 0x7fff9d7fe6c0 (LWP 238693)]
[New Thread 0x7fff9dfff6c0 (LWP 238694)]
[Thread 0x7fff9d7fe6c0 (LWP 238693) exited]
[Thread 0x7fff9dfff6c0 (LWP 238694) exited]
[Thread 0x7fffeaffd6c0 (LWP 238309) exited]
Cancel
/usr/include/c++/12.2.1/bits/stl_vector.h:1123: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = std::vector<SHAPE_LINE_CHAIN>; _Alloc = std::allocator<std::vector<SHAPE_LINE_CHAIN> >; reference = std::vector<SHAPE_LINE_CHAIN>&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.

Thread 39 "kicad" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffa73fc6c0 (LWP 238506)]
0x00007ffff56a08ec in ?? () from /usr/lib/libc.so.6
(gdb) bt -full
#0  0x00007ffff56a08ec in  () at /usr/lib/libc.so.6
#1  0x00007ffff5651ea8 in raise () at /usr/lib/libc.so.6
#2  0x00007ffff563b53d in abort () at /usr/lib/libc.so.6
#3  0x00007ffff58d30e2 in std::__glibcxx_assert_fail(char const*, int, char const*, char const*) (file=<optimized out>, line=<optimized out>, function=<optimized out>, condition=<optimized out>) at /usr/src/debug/gcc/gcc/libstdc++-v3/src/c++11/debug.cc:60
#4  0x00007fffba9b6021 in  () at /usr/bin/_pcbnew.kiface
#5  0x00007fffbb062506 in  () at /usr/bin/_pcbnew.kiface
#6  0x00007ffff58d72c3 in std::execute_native_thread_routine(void*) (__p=0x555556c5b6b0) at /usr/src/debug/gcc/gcc/libstdc++-v3/src/c++11/thread.cc:82
#7  0x00007ffff569ebb5 in  () at /usr/lib/libc.so.6
#8  0x00007ffff5720d90 in  () at /usr/lib/libc.so.6
```

RigoLigoRLC, 2023-03-25:
I've thoroughly debugged KiCad to find out the cause. Turns out that, when the plugin finished executing, the connectivity data inside KiCad is corrupted, and will cause an std::vector to read out of bounds.

To be precise, after the script exited and `PYTHON_ACTION_PLUGIN::Run` returned, a corrupted connectivity data, whose target polygon index is higher than the total number of polygons that the zone has, can be found (I wrote custom dbghelp functions to find that out, and leveraged SearchClusters function to quickly search through all of the items).
[image]

This is the exact same filtering code like the actual Segfault location. A bad via-to-zone connection is found.
[image]

`m_triangulatedPoly` in the first screenshot is a `std::shared_ptr<SHAPE_POLY_SET>     m_triangulatedPoly;`, so it meant the geometry of the fill zone has been unexpectedly changed before the editor could propagate the changes to connected items (or because the method used is not going to propagate the geometry changes). And after commenting out all the `self.RefillBoardAreas()`, the script won't crash KiCad anymore.

I suggest commenting out the board refill code to mitigate the issue.

waterlubber, 2023-04-17:
I've got a board that reproduces this issue, if you need it. I will need to remove some personal information if I upload it, so I'll only send it here if it's necessary.

jsreynaud, 2023-06-08:
Last commit fix that. For the moment the only solution is to avoid automatic refill...

commit 680af4ef (2023-06-08): Fix some crash due to refill zone. Fix #61
closed 2023-06-08 by jsreynaud