jsreynaud/kicad-action-scripts #64: exception: missing HitTestInsideZone: To Be Fixed

Rom4ik-glitch, 2023-05-04:
What to do with this exception and how to fix it

ahmad-kassem, 2023-05-06:
Me too, I am having the same problem.

ke5gdb, 2023-05-09:
I'm seeing the same issue on 0.3.0 for KiCAD v7.0. 

[image]

```
Application: KiCad PCB Editor x64 on x64

Version: 7.0.1, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 5.0.1
	FontConfig 2.14.1
	libcurl/7.83.1-DEV Schannel zlib/1.2.13

Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Mar 11 2023 03:41:46
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.80.0
	OCC: 7.6.2
	Curl: 7.83.1-DEV
	ngspice: 39
	Compiler: Visual C++ 1934 without C++ ABI

Build settings:
	KICAD_SPICE=ON
```

73 de KE5GDB

navig2712, 2023-05-17:
Same issue

readmodifywrite, 2023-05-31:
Same here.  Kicad 6.

This should not be on the official plugin list if it doesn't work.

jsreynaud, 2023-06-08:
Hi,

This exception should not avoid via placement. If there is no via, perhaps you have zones that are not on the same net. Or perhaps you don't select appropriate net in the dialog.
Please update with more details...

readmodifywrite, 2023-06-09:
In my case, it was with a 4 layer stackup.  The vias are for GND, and I have GND fills on layers 1, 3, and 4.  Layer 2 is a power plane layer.

I just tried it again, but with the power plane deleted.  Vias everywhere!

So that's the issue, it's avoiding other zones not on the same net.

I think in general, I would prefer vias to appear on the zone I've selected regardless of other zones: I'm ok with puncturing the plane layer, it's an intentional design choice.  But there might be other designers who don't want that behavior.  It makes sense when you flood the entire layer, but less so when you do specific planes in specific areas.

So the easy workaround is just remove the fills for the other nets, do the via stitching, and then put the fills back.  Easy if you do a flood on the entire layer, perhaps less so if you do targeted planes in specific areas.  But again to be fair, if you do planes in specific areas you might not want via stitching through them anyway.

Some possible compromise options:
1. Add an option to let us control that behavior.  A check box to allow/disallow puncturing zones on other nets.
2. Some kind of warning if not very many vias are placed (actual vias placed vs theoretical max given zone area and via spacing) to remind us to check if there is a conflicting zone (or even show a message indicating the conflicting zones).
3. Easiest: add some text in the dialog box that explains that vias won't be placed in any areas where there is a zone conflict.

Thanks!

jsreynaud, 2023-06-09:
> So that's the issue, it's avoiding other zones not on the same net.

Yes. As I see this is also the way of work on some others CAD softwares.
But it can be an option to allow this by pass.

Feel free to submit a pull request for that ;)

ahmad-kassem, 2023-06-10:
there has been an update for the plugin recently and after the update there is a new checkbox called (ignore areas on other layers), after selecting it the stitching works when you have another plane with different net, so for me the problem is solved.