jsreynaud/kicad-action-scripts #67: KiCad 7 Via Stitching Error

alcohen, 2023-11-26:
I installed Action Scripts into my KiCad instance (7.0.1). When I run Via Stitching, I get an error "Done. 0 vias placed. You have to refill all your pcb's areas/zones !!!" However it is freshly refilled.

Board passes DRC except for few islands in the copper pour. First layer is signal with copper pour to GND, second layer is GND plane.

I've attached a screen shot showing the error and details.
[image]

jsreynaud, 2023-11-27:
If you have a full layer/zone/area without GND as net, This plugin will not fil with via. This plugin automatically avoid area/zone when it's not in GND (or selected net).

alcohen, 2024-02-16:
It's been a while, but I realize that I forgot to thank you for your response. I'll try to use thus again on my next board.

closed 2024-03-25 by jsreynaud