jsreynaud/kicad-action-scripts #71: kicad8.0RC1.91 bug

HaydenHu, 2024-01-20:
[image]

Redherring32, 2024-01-20:
Can confirm this also happens for me, but only on pre existing files made in another version of Kicad.
If I create a new file, and add an edge cut rectangle and F.Cu fill, it will happily place vias.

HaydenHu, 2024-01-21:
> Can confirm this also happens for me, but only on pre existing files made in another version of Kicad. If I create a new file, and add an edge cut rectangle and F.Cu fill, it will happily place vias.

Isn't it supported to file on the project? It is true that new PCBs can add vias, but they will not avoid other net!

jsreynaud, 2024-01-21:
Branch v8.0 of this project handle kicad V8. Please be sure you use this version.

HaydenHu, 2024-01-21:
> 本项目的分支 v8.0 处理 kicad V8。请确保您使用此版本。

I've tried them all and have merged into the master

jsreynaud, 2024-01-21:
Ok, I had updated a modification. Could you confirm it's ok ?

HaydenHu, 2024-01-21:
> 好的，我已经更新了修改。你能确认没问题吗？

PCBs from both old and new projects will not report errors, but vias will not avoid other nets
[image]

HaydenHu, 2024-01-27:
> Ok, I had updated a modification. Could you confirm it's ok ?

！！？

Saatvik-Aggarwal, 2024-02-26:
I'm not able to get it to fill on a KiCad v8.0.0 project (upgraded from 7.0.3)--says 0 vias placed.

ekartgo, 2024-02-27:
Using the latest code with 8.0 release creates a grid without avoiding any pads or traces using the grid or star options.

Unfortunate as this is a very useful tool for my prototypes and I'd be willing to donate to get it fixed.

jsreynaud, 2024-02-28:
Let me check to fix this issue...

HaydenHu, 2024-03-04:
> Let me check to fix this issue...

Did you find the reason for the error?

HaydenHu, 2024-03-07:
If you have time, fix this bug, I can't do without it

lucblender, 2024-03-11:
Using kicad 0.8.0 and v8.0 branche. I am facing the same issue with vias overlapping traces.

VoxelTek, 2024-03-21:
I have this somewhat similar error:
[image]

VoxelTek, 2024-03-22:
It's worth noting that this error appears and nothing actually happens, no vias are placed. The project was started as a KiCad 8.99 project and has always been one.

jsreynaud, 2024-03-24:
I had push an update. Please retry with last commit (on V8.0 branch of course...)

VoxelTek, 2024-03-24:
I'm unsure as to the error everyone else is getting, but I still get this issue:

[image]

ekartgo, 2024-03-25:
Seems to work correctly with v8 production now. Thanks for getting this resolved!

frizzyversion, 2024-03-25:
Hello,

I am running KiCad 8.0.1 on Windows and using the latest commit on the v8.0 branch and I am experiencing this exact issue.

[image]

@ekartgo what version of KiCad are you using?

lucblender, 2024-03-25:
Kicad 8.0.1, I tested the commit [ada62f8](https://github.com/jsreynaud/kicad-action-scripts/commit/ada62f8c9b0f11092514945f5352e19447a3eb48) on branch v8.0. The bug seems fixed, via doesn't overlap with traces any more!

jsreynaud, 2024-03-25:
> Hello,
>
> I am running KiCad 8.0.1 on Windows and using the latest commit on the v8.0 branch and I am experiencing this exact issue. [...]

Could you double check that ? It's working for at least 2 others people ...

jsreynaud, 2024-03-25:
> I'm unsure as to the error everyone else is getting, but I still get this issue: [...]

I think your issue is a different problem. Please provide details on a separate issue.

frizzyversion, 2024-03-25:
@ekartgo @jsreynaud 

It looks like it is now resolved, I am not sure what I did wrong but I deleted the plugin folder, double-checked I was on the v8.0 branch, copied the "ViaStitching" folder into the scripts folder, and it is now working. Thank you for fixing this 👍 

[image]

commit ada62f8c (2024-03-24): Error in VECTOR2I create. Float is not a valid option for parameters... Should FIX #71
closed 2024-04-02 by jsreynaud