jsreynaud/kicad-action-scripts #74: release for KiCad 8

SirBramble, 2024-03-24:
More of a question than a issue. Is a release version for KiCad 8 planned? 

A couple of friends and me really like this plugin, but started migrating to KiCad 8. However we never quite managed to get the plugin installed correctly. It would be a lot easier if we could install it via the plugin manager in KiCad.

Not trying to push anything, just wanted to know if we can expect any releases in the future. And if not, a small guide on how to install it via git would be greatly appreciated.

jsreynaud, 2024-03-24:
Yes there is a bug on V8.0 branch (see #71). I had push an update on V8.0 branch. Please try and check.
For availability in the plugin manager. I'm expect the pluging is stable for V8.0 before plublish it. 
Anyway you can manually install it by git in first approch.

SirBramble, 2024-03-24:
ok, I got it to work.
I put the folders (Via stiching and Circular zone) directly in to scripting/plugins and updated from "tools->External Plugins->Refresh Plugins".
Tested it on a PCB created in KiCad 8 and one migrated from KiCad 7. Both work.

closed 2024-03-28 by jsreynaud