jsreynaud/kicad-action-scripts #75: Via stitching error in KiCad 8.99 Nightly

VoxelTek, 2024-03-26:
Hi there,

When trying to use Via stitching in KiCad 8.99, the nightly release, I get this error:

[image]

This is the version I am using:
[image]

Here is a video demonstrating the issue:
https://github.com/jsreynaud/kicad-action-scripts/assets/53562267/9223eedd-d8bb-4816-ad01-7a410bf37b15

Debug mode does not appear to help, as shown in the video.

I would greatly appreciate any assistance given regarding this error.

Kind regards,
@VoxelTek

VoxelTek, 2024-03-29:
Update:

I modified `FillArea.py`, and changed line 602 from:
```py
area_clearance = area.GetLocalClearance()
```
to:
```py
area_clearance = int(area.GetLocalClearance())
```

This solved the errors I was having previously, but now, I get this error:

[image]

jsreynaud, 2024-04-03:
Yes, there are some missing feature in dev branch of KiCad that avoid to have this plugging working for the moment.
Once KiCad will provide needed functions I'll update it....