jsreynaud/kicad-action-scripts #76: Via Stiching Generator ignores pads and tracks

Keilomatico, 2024-03-30:
Hi there,

first I wanted to thank you for the amazing pluging! I have been using it on almost all of my PCBs so far.
However, I cannot get it to consider tracks and pads using version 8.0.0 or 8.0.1 on Windows 11. It just places the vias on top (see screenshot). I get an exception with processing pads and tracks for each via that is being placed.

[image]

19:29:27: Via Stitching
19:29:28: Processing Target Area: GND, LayerName: F.Cu...
19:29:28: Processing all vias of target area...
19:29:28: Processing all pads...
19:29:28: exception on Processing all pads...
[...]
19:29:28: exception on Processing all pads...
19:29:28: Processing all tracks...
19:29:28: exception on Processing all tracks...
[...]
19:29:28: exception on Processing all tracks...
19:29:28: Processing all existing drawings...
19:29:28: Done. 45 vias placed. You have to refill all your pcb's areas/zones !!!

Application: KiCad PCB Editor x64 on x64

Version: 8.0.1, release build

Libraries:
	wxWidgets 3.2.4
	FreeType 2.12.1
	HarfBuzz 8.3.0
	FontConfig 2.14.2
	libcurl/8.5.0-DEV Schannel zlib/1.3

Platform: Windows 11 (build 22631), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Mar 15 2024 01:52:47
	wxWidgets: 3.2.4 (wchar_t,wx containers)
	Boost: 1.83.0
	OCC: 7.7.1
	Curl: 8.5.0-DEV
	ngspice: 42
	Compiler: Visual C++ 1936 without C++ ABI

jsreynaud, 2024-03-30:
Please double check that you use the v8.0 branch of the pluging. This bug
is resolved in v8.0 branch...

Keilomatico, 2024-03-31:
Ah, stupid me! Yes, I was not on the v8.0 branch but on master... Thanks!

closed 2024-03-31 by Keilomatico

gu50, 2024-06-02:
I am on v8.0.2, using v8.0.5 of the plugin:
[image]

Debug log is still showing exactly the above exceptions - it's actually never been working on v8 for me since updating. 

I tried it a couple of times and even manually dropping the files from the v8 branch into the plugin folder, which didn't change anything. 

@Keilomatico : is it really working for you?

Keilomatico, 2024-06-03:
Sorry to hear that! For me everything worked perfectly after switching to the v8 branch