jsreynaud/kicad-action-scripts #80: Via Stitching Generator ignores keepout areas

xobs, 2024-07-24:
The via stitching generator appears to ignore keepout areas.

For example, here is a U.Fl connector with a keepout area underneath it:

[image]

This array was generated with the defaults, except "Ignore areas on other layers" was selected because otherwise it generates 0 vias.

This is running kicad-action-scripts version 8.0.5 on kicad 8.0.4.

xobs, 2025-02-13:
Annoyingly, there's no way to manually remove the offending vias, since deleting it removes everything.

jsreynaud, 2025-02-13:
> Annoyingly, there's no way to manually remove the offending vias, since deleting it removes everything.

Yes you can. All vias are inside a group. So ungroup them or enter in group then you will be able to delete individuals vias.

xobs, 2025-02-13:
Oh, thank you! I didn't know that there was a workaround. I thought it generated a special kind of multi-via object that couldn't be manipulated, which is why the "Delete Vias" button just shows an alert.

Of course, once you ungroup things there's no way to undo the via stitching. Even so, that's a much better solution than to keep running the generator repeatedly while fiddling with the grid spacing in the hopes that I find a value that doesn't generate violations.