jsreynaud/kicad-action-scripts #81: Release for Kicad 9

vladkvit, 2025-02-26:
The plugin is currently not available for KiCad 9.

When I copy-paste the plugin folder manually into the 9.0/3rdparty/plugins folder, it seems to work just fine.

ozzysv, 2025-03-25:
This does not work properly on version 9. The plugin ignores contact pads of other networks and places vias on them, which leads to critical errors !!!

[test.zip](https://github.com/user-attachments/files/19447081/test.zip)

[image]

zodoczi, 2025-03-29:
is there any patch or workaround for this? thank you!

zodoczi, 2025-03-31:
> > is there any patch or workaround for this? thank you!
> 
> [#82](https://github.com/jsreynaud/kicad-action-scripts/pull/82)

Thank you! :) Great, I patched manually but now kicad not let me install the zip file, telling me not valid json file in it :D

kuwoyuki, 2025-04-01:
> > > is there any patch or workaround for this? thank you!
> > 
> > 
> > [#82](https://github.com/jsreynaud/kicad-action-scripts/pull/82)
> 
> Thank you! :) Great, I patched manually but now kicad not let me install the zip file, telling me not valid json file in it :D

You could just clone the repo into `~/.local/share/kicad/9.0/scripting/plugins` (on Windows `%APPDATA%\kicad\scripting\plugins`) and apply the patch:

```diff
diff --git a/ViaStitching/FillArea.py b/ViaStitching/FillArea.py
index f81113b..b3a29e0 100644
--- a/ViaStitching/FillArea.py
+++ b/ViaStitching/FillArea.py
@@ -696,7 +696,7 @@ STEP         = '-'
                                 m.SetViaType(VIATYPE_THROUGH)
                                 m.SetDrill(int(self.drill))
                                 m.SetWidth(int(self.size))
-                                if pad.GetEffectivePolygon().Collide(m.GetEffectiveShape()):
+                                if pad.GetEffectivePolygon(pad.GetLayer()).Collide(m.GetEffectiveShape()):
                                     rectangle[x][y] = self.REASON_PAD
 
                     except:
```

```sh
~/.local/share/kicad/9.0/scripting/plugins/kicad-action-scripts $ git apply alayer.patch
```

zodoczi, 2025-04-01:
thanks, but there is no more scripting folder anymore since 9.0, looks like

[image]

And the path is: AppData\Roaming\kicad\9.0

kuwoyuki, 2025-04-01:
https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/README.md#git
Follow this guide put replace the git clone URL, also note:
> You may need to create the `scripting/plugins` folder if it does not exist.

zodoczi, 2025-04-01:
> https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/README.md#git Follow this guide put replace the git clone URL, also note:
> 
> > You may need to create the `scripting/plugins` folder if it does not exist.

thank you very much, but this is not for Kicad 9 but 8.99, I see no more plugins sub-folder in Kicad 9.0 folder structure (by default)

kuwoyuki, 2025-04-01:
Just create the `kicad/9.0/scripting/plugins` directory. I'm using KiCad nightly and PCBNew does read that plugin directory and loads the plugins fine, it's just not created automatically.

ajaybhargav, 2025-09-08:
simplest way to install is:

1. download the plugin from release and extract the zip.
2. edit the files mentioned in: https://github.com/jsreynaud/kicad-action-scripts/issues/81#issuecomment-2768624688
3. repack the zip and install.

this way when new version comes its easy to update it.

jsreynaud, 2025-09-09:
Hi all. I finally merge this pull request. Update of metadata in KiCad is submitted. Should be up-to-date in few days in KiCad...

closed 2025-09-09 by jsreynaud